riscv/riscv-memory-tagging #4: CSR bit positions

SiFiveHolland, 2024-11-01:
Since Zimte depends on XLEN >= 64, Smpnpm, and Ssnpm, it should use CSR bits in the high half of the `envcfg` CSRs, like pointer masking does. (The register diagrams should be updated to include the PMM field.)

Also, why is there no `MT_ASYNC` field in `senvcfg`? It would be supervisor software's policy whether or not it allows asynchronous reporting in U-mode.

deepak0414, 2025-01-01:
Yeah good point. I'll revise to add CSR bits in high half.

> Also, why is there no MT_ASYNC field in senvcfg? It would be supervisor software's policy whether or not it allows asynchronous reporting in U-mode.

A bug. yeah thanks for pointing it out. Seems like I missed it.

deepak0414, 2025-03-10:
This is fixed in spec
per- https://github.com/riscv/riscv-memory-tagging/pull/11

closed 2025-03-10 by deepak0414