riscv/riscv-memory-tagging #6: CBO.ZERO and MTE

Adam-pi3, 2024-11-05:
CBO.ZERO may work as a STORE operation. We should clarify that MTE is applicable to CMO (especially CBO.ZERO operation).

Adam-pi3, 2025-03-10:
The following PR should address this issue:
https://github.com/riscv/riscv-memory-tagging/pull/28

Please let me know if any changes are needed.

deepak0414, 2025-07-01:
Based on the meeting on 7/1/2025, Adam will send out revised PR removing wordings for cbo.zero and point to unpriv spec part of pointer masking spec illustrating loads/store types.

Adam-pi3, 2025-07-03:
Done. Please let me know if the wording is correct or should be changed.

deepak0414, 2025-08-12:
Closing this issue.
Addressed in https://github.com/riscv/riscv-memory-tagging/pull/63

closed 2025-08-12 by deepak0414