riscv/riscv-memory-tagging #8: LOWEST_VADDR_CURR_PRIV/HIGHEST_VADDR_CURR_PRIV

janhoogerbrugge-nxp, 2024-12-10:
I was wondering where these values are coming from:

LOWEST_VADDR_CURR_PRIV: Lowest virtual address in current privilege.
HIGHEST_VADDR_CURR_PRIV: Highest virtual address in current privilege.

Are these values 0 and 2^64-1 or something else? If something else, how does the core get them?

Jan.

deepak0414, 2025-01-01:
Sorry for being late on this.

Since memory tagging is 64 bit only feature.
Planning to make a dependency of `Svatag` on `Svukte` (https://github.com/riscv/riscv-isa-manual/pull/1564)

`Svukte` makes it implicit that user and supervisor are split half evenly.

deepak0414, 2025-05-06:
https://github.com/riscv/riscv-memory-tagging/pull/32