riscv/riscv-memory-tagging #9: gentag and invalid tags

janhoogerbrugge-nxp, 2024-12-11:
Should we prevent that the random number tag generation generates the invalid tag for pointer_tag_width/mc_tag_width = 4?

deepak0414, 2025-01-01:
Yes I'll add that as clarification note in spec so that its clear to implementers.

janhoogerbrugge-nxp, 2025-01-06:
I have a few remarks / questions related to this (based on the document from a few days ago):

- I see the notation rs1[b63:pointer_tag_width]. Shouldn't this be rs1[b63:b63-pointer_tag_width+1]?
- For gentag I see read that rd is cleared before the tag bits are set. I don't see this for the description of addtag. Should that also start with clearing rd?
- You might what to mention that in the function epilog t0 is used to restore the tag in the stack pointer.
- Gentag should generate a tag such that rd xor rs1 is not the invalid tag
- Addtag should generate a tag such that rd + rs1 is not the invalid tag
- Is it possible that addtag also copies the address so that the addi that follows in the example is not needed?

deepak0414, 2025-01-07:
>I see the notation rs1[b63:pointer_tag_width]. Shouldn't this be rs1[b63:b63-pointer_tag_width+1]?

I as using `bit_position:length` notation. But I see that riscv priv and unpriv spec use absolute position notation. Will fix it.

> For gentag I see read that rd is cleared before the tag bits are set. I don't see this for the description of addtag. Should that also start with clearing rd?

Yeah good catch. Will fix it.

> You might what to mention that in the function epilog t0 is used to restore the tag in the stack pointer.

Yeah one more good point. But instead of `t0`, I think example should use something like frame pointer (s0).
`t0` is temp according to calling convention and will be lost across function calls. Using `s0` should be more
useful. Will add that.

> Gentag should generate a tag such that rd xor rs1 is not the invalid tag

Yes, will fix it.

> Addtag should generate a tag such that rd + rs1 is not the invalid tag

Yes, will fix it.

> Is it possible that addtag also copies the address so that the addi that follows in the example is not needed?

No this is not possible for the same reason we had to split tag generation `gentag` + `xor`
If memory tagging is disabled, then rd will be zero. Thus NULL pointer will get generated for stack objects.

deepak0414, 2025-03-24:
We don't have `invalid` mc_tag or special meaning to `zero` pointer_tag anymore.
Closing this issue.

closed 2025-03-24 by deepak0414