riscv/riscv-memory-tagging #25: GENTAG/ADDTAG generating zero-tags

cmuellner, 2025-02-13:
Zero-tags have a special meaning as they are the default tags:
* an unset memory tag is expected to be zero
* an unset pointer tag is zero

This raises the question of whether GENTAG/ADDTAG should generate a zero-tag as this leads to tagged-memory being potentially equally tagged to non-tagged-memory.

I would prefer to have tagged memory never be compatible with zero-tagged pointers.
This could be realised by assuring that GENTAG does not generate zero-tags and ADDTAG would skip zero-tags.

deepak0414, 2025-03-10:
Tag elision mechanism is still under discussion.
As per discussions in our last meetings, we won't choose pointer_tag = 0 as elision mechanism.
Keeping the issue open. Will close when we settle on mechanism.

deepak0414, 2025-03-24:
We def won't use `pointer_tag=0` for elision. Closing this issue.

closed 2025-03-24 by deepak0414