riscv/riscv-memory-tagging #60: `settag` should not be able to raise misaligned

SiFiveHolland, 2025-07-29:
The spec mandates all implementations support `pointer_tag_width == 4`, which requires that the hardware can perform a RMW sequence to update tags in memory. This means hardware must be able to serialize `settag`  operations at overlapping addresses. This means hardware _should_ be able to treat a `settag` of multiple tags as a RMW of a larger, naturally-aligned word. And any RV32 implementations, as well as RV64 implementations that support `pointer_tag_width == 7`, must be able to split a `settag` operation into multiple smaller memory writes.

Given the addresses involved in stack tagging depend on the `sp` at function entry, which is only aligned to 16 bytes (one `mc`), it isn't feasible to require the address given to `settag` to be naturally aligned. This would effectively require all stack objects to themselves be naturally aligned, which would waste memory.

So I think we should require hardware to support arbitrary combinations of `rs1` alignment and `chunk_count`.

deepak0414, 2025-07-29:
There is pointer masking dependency and thus applies only to RV64 (not to RV32). `settag` accesses can be misaligned. Do we really want to force the behavior misaligned can't happen? I believe SoC owner can handle that PMAs, right?

> So I think we should require hardware to support arbitrary combinations of rs1 alignment and chunk_count.

spec mentions that settag rs1 will align input address to 16 byte boundary and then calculate tag va.

deepak0414, 2025-08-13:
Please take a look at this PR
https://github.com/riscv/riscv-memory-tagging/pull/69

deepak0414, 2025-09-24:
Merged