riscv/riscv-memory-tagging #72: Async reporting : add wording for resolving all tag checks on privilege transitions

deepak0414, 2025-09-09:


lfiolhais, 2025-09-24:
I thought a bit more about the async mode and I believe there is a case we didn't discuss.

It is possible for a uarch to trigger a tag exception before the store is committed. In this instance, SW will get an exception before the store is architecturally visible. Even worse, to my understanding, it is possible to get an exception for a store that will never commit.

I think it's safe to say that the async mode wasn't designed with this behavior in mind.

Would it be possible to add a normative statement that mandatesv that the exception is only visible _after_ the corresponding store commits? I'd propose: "In asynchronous mode, tag exceptions on stores are only reported after their corresponding stores are architecturally visible."

deepak0414, 2025-10-30:
Asynchronous mode is removed from spec
See PR https://github.com/riscv/riscv-memory-tagging/pull/80

closed 2025-10-30 by deepak0414